Open-Smartwatch/open-smartwatch-os #275: Light sleep and really really long button press bug

Gladox114, 2022-07-24:
So when enabling DO_LIGHT_SLEEP then there is a bug using the "Really really long button press".

Whats the problem?
When trying to light sleep over this button press then the next App gets cycled.

A Showcase of this problem.
https://imgur.com/a/aDOuWCa

How to reproduce?
Get the newest version.
enable DO_LIGHT_SLEEP in config or in the web interface.
try to light sleep using the "Really really long button press"
Notice how you switched the app which shouldn't happen

I tried to fix that and made a Pull request but I get misunderstood because **this has nothing to do with being able** to enable or disable light sleep.  #272

RuffaloLavoisier, 2022-07-24:
@simonmicro It's a very interesting. Could it be a hardware problem?

Gladox114, 2022-07-24:
@RuffaloLavoisier 
Could you not recreate it?

I mean I tried to follow the code and in my understanding. The code allows it to do that.

Following the code from [switcher.cpp in its loop](https://github.com/Open-Smartwatch/open-smartwatch-os/blob/bc08af49ac33bde2c81f463ee0a1be0812b8d388/src/apps/main/switcher.cpp#L23), we can see that _doSleep will be true when _btn aka BUTTON_1 is held down long enough.

Then a switch case on LONG_PRESS there is a [if statement](https://github.com/Open-Smartwatch/open-smartwatch-os/blob/bc08af49ac33bde2c81f463ee0a1be0812b8d388/src/apps/main/switcher.cpp#L31) that sets _doSwitch to true if BUTTON_1 was held long enough.

Those two will definitely trigger and set _doSleep and _doSwitch to true if BUTTON_1 is held long enough.

Meaning that in the final [if statement](https://github.com/Open-Smartwatch/open-smartwatch-os/blob/bc08af49ac33bde2c81f463ee0a1be0812b8d388/src/apps/main/switcher.cpp#L43) the light sleep gets activated. Because it isn't a deep sleep the code continues and triggers the if statement with _doSwitch which cycles the app.

simonmicro, 2022-07-24:
Hey, thanks for the bug report - no, @RuffaloLavoisier this is not a hardware bug. As @Gladox114 explained pretty nice, its a "feature" inside the app switcher.

@RuffaloLavoisier would you care to patch this? I'm pretty full atm and won't be able to resolve this until the week after next week.

@Gladox114 Thanks again - sorry for closing the first issue, I think there was a miscommunication 😅

RuffaloLavoisier, 2022-07-24:
@Gladox114 I will review it within this week. Re-open #272

RuffaloLavoisier, 2022-07-25:
@Gladox114 
Merged, Thank you for your contribution. ! To 1bc59a7b381165ac094aee8aaebdcccb8d59bacc

closed 2022-07-25 by RuffaloLavoisier