Open-Smartwatch/open-smartwatch-os #300: String.h conflicts with string.h

ameharoo, 2022-09-28:
# What is not working?
The emulator does not compile due to case-insensitive filesystem on Windows. Conflict between string.h and [String.h](https://github.com/Open-Smartwatch/open-smartwatch-os/blob/master/emulator/include/String.h).
in file `emulator/lib/ArduinoJson/src/ArduinoJson/Strings/Adapters/RamString.hpp:26:21`

# How to reproduce this
Steps to reproduce the behavior:
1. Build emulator.run target by MSYS MinGW on Windows 10 21H2.
2. See error

# What should happen instead?
Including string.h instead String.h.

# Environment
- **Model** Does not matter.
- **OSW-OS** 0a71a6d85cd18bba1362f524b6713dfb84047a7e
- **Browser** Does not matter.

# Additional context
Maybe change the name of String.h to something else?

simonmicro, 2022-09-29:
So the problem is that we want to include `string.h`, but Windows instead includes `String.h`? Could you post the build error?

Problem is, that the emulator MUST overwrite the Arduino-File `String.h` with its custom implementation and well, the name is therefore fixed... Maybe you can add a flag to be case sensitive?

simonmicro, 2022-10-11:
Closing this issue, as I need at least some logs, otherwise I can't do anything...

closed 2022-10-11 by simonmicro