Open-Smartwatch/open-smartwatch-os #307: NTP time updating doesn't work due to time drift compensation.

FranseFrikandel, 2022-10-07:
# What is not working?
PR #263 appears to break NTP network update. The time is not updated when pressing the update button when connected to WiFi.

It appears that immediatly as the time is set, the automatic time drift detection pulls the time back from the onboard RTC, prevent time from being effectively updated. Commenting out the code that keeps the ESP32 internal clock in sync appears to fix the issue as well (line 42 to line 50 in src/devices/esp32.cpp).

# How to reproduce this
Try to update the time using NTP when it is significantly off.

# What should happen instead?
The time should update.

# Environment
- OSW Light v3.2
- OSW latest main branch

# Additional context
Serial monitor:
```
src/services/OswServiceTaskWiFi.cpp: [NTP] Started update...
src/devices/esp32.cpp: Resynced internal ESP32 clock with other provider due to significant time difference (> 4 seconds).
src/services/OswServiceTaskWiFi.cpp: [NTP] Update finished (time of 1665131972)!
```

simonmicro, 2022-10-09:
Okay. That's the bug! I noted it, but not cared to investigate... My bad!

closed 2022-10-10 by RuffaloLavoisier

simonmicro, 2022-10-11:
@RuffaloLavoisier No need to close fixed issues manually -> https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue

This would allow users to see what is not fixed on `master` and by following the issue-history see potential fixes on `develop`.