Open-Smartwatch/open-smartwatch-os #340: Add integration tests

akmal-threepointsix, 2023-01-06:
There is already a decently working emulator for the OSW, so it would be great to add some UI tests, which automatically:

1. Start emulator
2. Perform some actions (e.g. press some buttons in the given order)
3. Compare obtained results with the expected one

For example, integration test for the stopwatch can be:

1. Start emulator
2. Press buttons in order to switch emulator to stopwatch app and start it
3. After 1 second, stopwatch time must be greater or equal than 1 (e.g. `EXPECT(totalTimeSec >= 1)`)

OSW emulator uses [imgui](https://github.com/ocornut/imgui) graphical library. This library has a [testing engine](https://github.com/ocornut/imgui_test_engine). Example of this engine:

https://user-images.githubusercontent.com/8225057/182409619-cd3bf990-b383-4a6c-a6ba-c5afe7557d6c.mp4

akmal-threepointsix, 2023-01-06:
- Issue is open for the discussion
- If you can, please assign it to me, since I am working on testing the app

simonmicro, 2024-04-06:
@akmal-threepointsix Are you still on this?

akmal-threepointsix, 2024-04-06:
> [simonmicro](https://github.com/simonmicro) linked a pull request [on Feb 18, 2023](https://github.com/Open-Smartwatch/open-smartwatch-os/issues/340#event-8553428764) that will close this issue

Hello, this issue was closed: https://github.com/Open-Smartwatch/open-smartwatch-os/pull/351, https://github.com/Open-Smartwatch/open-smartwatch-os/pull/367

closed 2024-04-06 by akmal-threepointsix