Open-Smartwatch/open-smartwatch-os #342: Create the notification system

cutieskye, 2023-01-07:
The OSW-OS already has applications that could notify users. For example, users could see a notification when they reach their daily walking goal. They could also be informed about the weather forecast. 

Creating the alarm or the stopwatch is pointless without having the means to deliver notifications.

I propose developing the notification system as a background service that will display messages the apps may send to the user. The scheduler should handle all messages uniformly regardless of which app has published them. Its responsibilities are calculating the time when a notification should fire and drawing the appropriate banners on the screen.

Of course, this notification system will only be complete with some hardware components, like a buzzer or vibration motor.

cutieskye, 2023-01-08:
This issue should be linked to the pull request bearing the same name and pull requests Feature/alarm and Feature/timer.

simonmicro, 2023-01-09:
> hardware components

I could imagine, that we extend the provider-pattern we already have for the sensors to have providers to alert / vibrate the user. Maybe more generic like `requestAttention(level)`, which could be interpreted as a blinking LED, vibrator pattern or buzzer patterns :)

cutieskye, 2023-01-10:
https://user-images.githubusercontent.com/68032402/211615168-0d6509b2-ff8a-4df3-bb55-adc45bd39898.mov

Here's a short demo prepared by @akmal-threepointsix

closed 2023-03-28 by simonmicro