Open-Smartwatch/open-smartwatch-os #360: 3rd Party Watch Support (Watch, T-Watch)

simonmicro, 2023-04-02:
With this issue we will make the OSW-OS more portable to finally also run on:

- Watchy: https://github.com/sqfmi/Watchy
- T-Watch (2021 V1): https://de.aliexpress.com/item/1005003140780271.html
- T-Watch (2020 V3): https://de.aliexpress.com/item/1005002053650442.html
- T-Watch (2020 V2 GPS): https://de.aliexpress.com/item/1005002264354524.html

_The support for those devices will be officially "inofficial", meaning the user may need to port / extend it himself, whenever the hardware of these models change._

If you have any experience with those devices, or if you could support this (costly) effort, please come into our Discord (and ping @simonmicro ).

a-allard, 2023-10-18:
Out of curiosity, are you planning to update the OSW hardware at all?  From the website it appears there is/was a rev 4 and rev 5 in the works are one time.  Is that still true?  Or is that wishful thinking?

As noted [here](https://github.com/Open-Smartwatch/open-smartwatch-gps/issues/10) I was thinking of helping with an update.  Just need some indication of what changes would be helpful/beneficial.

Let me know if that would be helpful.  I really appreciate what you guys are doing setting up open source HW and SW for wearables.

Cheers!

simonmicro, 2023-10-18:
I plan to do it whenever I have time. For now, I work on this issue, but whenever I have time :wink: .

No seriously, there is still progress on OSW, although not as much anymore as we are all having a live (and not many others are here to help out). The LEv5 can be considered cancelled, as @uvwxy aborted development and never released the progress he made (so we have to start from scratch - so with time comes LEv6).

So, if you have time and want to help - go ahead!