Open-Smartwatch/open-smartwatch-os #362: Using GIF_BG flag makes the project to not compile.

HugoPlacer, 2023-04-05:
# What is not working?
The usage of GIF_BG drop a bunch of errors when compiling.

`/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/games/snake_game.cpp: In member function 'virtual void OswAppSnakeGame::setup()':
/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/games/snake_game.cpp:13:5: error: 'bgGif' was not declared in this scope
   13 |     bgGif->setup(hal);
      |     ^~~~~
/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/games/snake_game.cpp:13:18: error: 'hal' was not declared in this scope
   13 |     bgGif->setup(hal);
      |                  ^~~
/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/games/snake_game.cpp: In member function 'virtual void OswAppSnakeGame::stop()':
/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/games/snake_game.cpp:25:5: error: 'bgGif' was not declared in this scope
   25 |     bgGif->stop();
      |     ^~~~~
/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/games/brick_breaker.cpp: In member function 'virtual void OswAppBrickBreaker::stop()':
/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/games/brick_breaker.cpp:20:5: error: 'bgGif' was not declared in this scope
   20 |     bgGif->stop(hal);
      |     ^~~~~
/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/games/brick_breaker.cpp:20:17: error: 'hal' was not declared in this scope
   20 |     bgGif->stop(hal);

/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/watchfaces/OswAppWatchface.cpp: In member function 'virtual void OswAppWatchface::onStart()':
/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/watchfaces/OswAppWatchface.cpp:131:17: error: no matching function for call to 'OswAppGifPlayer::setup(OswAppV2::OswHalProxy&)'
  131 |     bgGif->setup(hal);
      |     ~~~~~~~~~~~~^~~~~
In file included from /home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/watchfaces/OswAppWatchface.cpp:11:
/home/hugo/Documentos/GitHub/open-smartwatch-os/./include/./apps/_experiments/gif_player.h:11:18: note: candidate: 'virtual void OswAppGifPlayer::setup()'
   11 |     virtual void setup() override;
      |                  ^~~~~
/home/hugo/Documentos/GitHub/open-smartwatch-os/./include/./apps/_experiments/gif_player.h:11:18: note:   candidate expects 0 arguments, 1 provided
/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/watchfaces/OswAppWatchface.cpp: In member function 'virtual void OswAppWatchface::onDraw()':
/home/hugo/Documentos/GitHub/open-smartwatch-os/src/apps/watchfaces/OswAppWatchface.cpp:162:16: error: no matching function for call to 'OswAppGifPlayer::loop(OswHal*)'
  162 |     bgGif->loop(OswHal::getInstance());`

# How to reproduce this
Steps to reproduce the behavior:
1. Go to 'CMakeLists.txt'.
2. set the flag GIF_BG in "target_compile_definitions".
3. Compile the emulator.
4. See error.

# What should happen instead?
Compile and show the GIF as background.

# Environment
- **Model** [Emulator]
- **OSW-OS** [I tried master and NewAppClassAndMore branch]

# Additional context
Nothing.

simonmicro, 2023-04-14:
1. Some games had copy-pasted broken code for that...
2. Also some watchfaces tried to use it with hal references (we changed that a long time ago)
3. The currently included GIF is way too big.........

simonmicro, 2023-04-14:
https://user-images.githubusercontent.com/24966116/232142296-3601cc2b-d66a-408c-a87c-8fa8171b3d61.mp4

It works now... More bad than good - it constantly crashes due to stack violations, but this is a problem for another day (and somebody else). I assume @uvwxy hacks to get the rendering working are too hacky...

PR #363 (2023-04-22): Bug/gif bg flag
Fixes #362 

I also needed to redesign the workflows to work with multiple flags defined at the same time, as the build-script was not proper cleaning all defines before (which caused size-problems with the `GIF_BG` flag). After changing that, too many flags were removed, so I re-engineered the flag-extraction from the docs to just contain all required ones.
PR #364 (2023-06-04): Next Release: Minor :tada:
Fixes #362
closed 2023-06-04 by simonmicro