Open-Smartwatch/open-smartwatch-os #365: Fix typo on Docs/Project

RuffaloLavoisier, 2023-06-04:
There are many typos in the project. This needs to be corrected.

simonmicro, 2023-06-04:
You mentioned the use of `codespell` to search for those... Maybe we should also add hat to the CI?

simonmicro, 2023-06-05:
I've unassigned myself, as I currently don't have the capacity to also work on this. If this remains a TODO for a while, I'll my add it to a milestone and take it over.

RuffaloLavoisier, 2023-06-05:
> You mentioned the use of `codespell` to search for those... Maybe we should also add hat to the CI?

A double check is required to validate the results through CI. It's a wonderful tool, but it's just a tool. It is up to humans to double-check the results of the modification.

commit c7476066 (2024-05-20): Close #365, fix typo
PR #390 (2024-05-20): Fix typo on root project
Close #365.
Fix some typo, please check it.
- exclude libraries
closed 2024-09-18 by RuffaloLavoisier