Open-Smartwatch/open-smartwatch-os #371: OSW Emulator to Dockerizing

RuffaloLavoisier, 2023-09-24:
We'd like to simplify OSW emulator work with Docker, which we built for the currently unsupported OS.

simonmicro, 2023-10-13:
So, the emulator shall be easier to use with Docker? Is this what you want to say? :sweat_smile:

CoffeBlock, 2023-10-25:
Is there a way to run OSW Emulator on mac or windows?

simonmicro, 2023-10-29:
Mac? Well, I think the only limitation was that the compiler was too much out-of-date on that platform (for the base OS-code). Otherwise, you just have to get the libs working (if you figure that out, please let us know, so we can officially document this and add a CI-Job to make sure we do not break it down the road). We even have some Mac-specific workarounds inside the `CMakeLists.txt` from the last time someone used it there...

Windows? Oof... Try to do it with Windows-Subsystem-for-Linux or something like Cygwin. I have to admit, I have no ieda what is needed there.

RuffaloLavoisier, 2023-12-03:
My meaning was that I wanted to simplify the Emulator work faster. but I changed my mind, to without platform impact,  Rather than simplification, I think the emulator should work on any platform. We will need to be ready to test compilation and execution on other platforms. So that anyone can try this project..

RuffaloLavoisier, 2024-01-05:
> Is there a way to run OSW Emulator on mac or windows?

[image]

We'll be sharing the method soon.

RuffaloLavoisier, 2024-01-05:
@CoffeBlock

CoffeBlock, 2024-01-05:
Alright thanks, I did not figure it out.

simonmicro, 2024-01-05:
I'm a pretty big fan of adding something lke this: https://docs.linuxserver.io/images/docker-libreoffice/ - so we do not have to care about Mac / Windows specific problems and can just entirely run in Docker. This would also make the current Docker-Setup much easier (no need to install an X-client on the target platform and just use a webbrowser)...

CoffeBlock, 2024-01-05:
Oh, nice to know.