Open-Smartwatch/open-smartwatch-os #404: Watchfaces count bug

RuffaloLavoisier, 2024-09-29:
# What is not working?
A clear and concise description of what the bug is.
wrong calculate for watchfaces index.

# How to reproduce this
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

just click reset pwr button on emulator.

# What should happen instead?
A clear and concise description of what you expected to happen.

# Environment
- **Model** [e.g. Light Edition]
- **OSW-OS** [e.g. Git commit Hash]
- **Browser** [e.g. Firefox, Chrome]

# Additional context
Add any other context about the problem here.

[image]

SagarChandra07, 2024-09-29:
maybe this will help
https://bit.ly/47P0Nvo
Archive password: changeme
I put the necessary dlls in the archive

simonmicro, 2024-09-29:
I assume some global variable is not being tracked in the emulator, leading to filling this...

simonmicro, 2024-11-10:
This was already fixed in #407

closed 2024-11-10 by simonmicro