Open-Smartwatch/open-smartwatch-os #411: getStepsOnDay bug

RuffaloLavoisier, 2024-10-12:
https://github.com/Open-Smartwatch/open-smartwatch-os/blob/fecfd0a1ed7eb580c20f2bc463a1fe1ba1e427c0/src/hal/environment.cpp#L197

Initially, the code was designed to print the data when the user requested the 'lastWeek' flag. However, upon further inspection, I noticed that when 'lastWeek' is enabled, no data is transferred if the 'dayOfWeek' and the requested 'weekDay' do not match. Therefore, the 'lastWeek' setting is valid only when the requested 'dayOfWeek' aligns with the current 'weekDay'.

I believe it's more appropriate to replace only the 'getStepsToday' data or the '_StepsCache' when 'lastWeek' is set, while keeping the other data unchanged and passing all the relevant information.

simonmicro, 2024-10-14:
Hi,

I think you misinterpreted the function in conjunction with the rest of the code: If you request the current day, it will be read from the sensor itself. If you request other days of the week, it will be read from the cache. Only exception is the current day and accessing last week: Because that data is still being held in the cache and not overwritten (as for the same day the data is always read from the sensor), this is the special case where you can still access that... Otherwise the function will report `0`.

RuffaloLavoisier, 2024-10-17:
This is issue is resolved.

closed 2024-10-17 by RuffaloLavoisier