Open-Smartwatch/open-smartwatch-os #413: Open-smartwatch device name standardization

RuffaloLavoisier, 2024-10-30:
https://github.com/Open-Smartwatch/open-smartwatch-os/blob/40ed6d66d2b42437c3f9b6ebed0cf63bfda4e3b6/include/config_defaults.h#L47
https://github.com/Open-Smartwatch/open-smartwatch-os/blob/40ed6d66d2b42437c3f9b6ebed0cf63bfda4e3b6/include/config.h.example#L17

The app recognizes the device name, but it must be unified with prefixes of the same specification.

Also, is the device name up to 8 characters.

simonmicro, 2024-11-02:
...are you talking about recognition in Gadgetbridge?

RuffaloLavoisier, 2024-11-04:
Yes, that’s correct. Although I’m unsure how things might change in the distant future, for now, it seems sufficient to modify this by attaching it as a description in the section where we set up web interfaces or device names.

simonmicro, 2025-01-18:
Maybe we can even allow longer device names, by implementing device scan responses... There is a promising discussion there: https://www.esp32.com/viewtopic.php?t=16239