Open-Smartwatch/open-smartwatch-os #414: Toast message service is not working

RuffaloLavoisier, 2024-11-08:
# What is not working?
A clear and concise description of what the bug is.

The floating message is not output.

# How to reproduce this
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

connect BLE . and send message to OSW device.

# What should happen instead?
A clear and concise description of what you expected to happen.

The floating message must be show on display.

# Environment
- **Model** [e.g. Light Edition] light_edition 3.3
- **OSW-OS** [e.g. Git commit Hash] (830d3054d2076de19aab88a0a9ef2dfa9ebe6c94)
- **Browser** [e.g. Firefox, Chrome] -

# Additional context
Add any other context about the problem here.

simonmicro, 2024-11-10:
It seems that my machine is working fine, while @RuffaloLavoisier Mac produces a binary with different behavior - even with the same commit hash AND same PlatformIO packages.

Let's see, if the artifacts of the pipeline have the same problem: https://github.com/Open-Smartwatch/open-smartwatch-os/actions/runs/11766491951

simonmicro, 2024-11-10:
@RuffaloLavoisier Try to clear everything regarding PlatformIO and caches from your system: Remove your `~/.pio` and `~/.platformio` folders, remove the `.pio` folder from the project and perform a clean build. If this still does not help, please upload a firmware, which has this issue here for further analysis.

RuffaloLavoisier, 2024-11-11:
@simonmicro 

In my side, the same issue happened with macOS workflow..

What do you think is the cause of this issue based on your review? Also, what should the correct values be under normal circumstances?

```
void OswServiceTaskNotifier::loop() {
    const std::lock_guard<std::mutex> lock{mutlimapMutex};
    auto utcTime = std::chrono::system_clock::from_time_t(OswHal::getInstance()->getUTCTime());
    auto currentTime = utcTime + std::chrono::seconds{static_cast<int>(OswHal::getInstance()->getTimezoneOffsetPrimary())};
    if (auto it = scheduler.begin();
            it != scheduler.end() && currentTime >= it->first) { . . . }
}
```

I believe it is a problem if a single piece of code is compiled and runs differently depending on the operating system.