Open-Smartwatch/open-smartwatch-os #422: Few issues/few enhancement requests

DuckersMcQuack, 2025-01-10:
# What is not working?
A clear and concise description of what the bug is.

# How to reproduce this
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Issues:
whenever watch turns on from sleep to activate screen, it emits a bright white flash before watch face has loaded.
I managed to do a button press that zoomed in the app drawer, and now unable to revert to stock
Wifi connects, but unable to adjust time.
If i set firmware to dd/mm/yyyy as i am used to, watch will not boot up at all after flashing. Just bootloop.

Feature request:

Dedicated setting page to adjust how much force is needed in miligram times sensitivity to activate "wake by wrist movement". As default wakes the watch with the slightest movement.
As well as settings for whether to toggle "wake by wrist movement and wake by tapping"

Documentation on default key binds for zooming in/out with buttons, increase/decrease brightness, set watch face as default. As those are missing.
A few links on the readme leads to nowhere/git.io page's 404 error page.

# Environment
- **Model** [e.g. Light Edition]
Light Edition
- **OSW-OS** [e.g. Git commit Hash]
[v0.17.2 /#418
- **Browser** [e.g. Firefox, Chrome]
chrome

# Additional context
Add any other context about the problem here.

simonmicro, 2025-01-11:
Okay, let's give you a quick feedback about all the different points you mentioned:

> 1: whenever watch turns on from sleep to activate screen, it emits a bright white flash before watch face has loaded.

Bug. The backlight is turned on before the first frame is rendered. Will fix / update the underlying library (which was breaking stuff before, hence we stuck to an older, buggy version).

> 2: I managed to do a button press that zoomed in the app drawer, and now unable to revert to stock

I have no idea what you mean with "zoomed in". If you want to return to home, you have to re-select the watchface from before, or reboot the watch.

> 3: Wifi connects, but unable to adjust time automatically. Doesn't update.

You have to enable automatic NTP upon wifi connection. Maybe there is some issue with NTP on your network? I had much success by connecting to the hotspot on my phone for this.

> 4: If i set firmware to dd/mm/yyyy as i am used to, watch will not boot up at all after flashing. Just bootloop.

Ooops, IF I can reproduce this, that's a nasty bug. Will fix and add a unit test to prevent that in the future. Keep in mind that this may only occur on a specific watchface and hence you did not include all needed information I may not be able to investigate this.

> 5: If i make adjustment attempts to the code, and re-flash, the welcome screen will not occur.

Yes. The "tutorial" will only be shown if the git-commit hash changes. This was done to prevent annoying developers by always showing it upon reflash. Just commit something and the next build will show it again.

> Documentation on default key binds for zooming in/out with buttons, increase/decrease brightness, set watch face as default. As those are missing.

Oh. Yes. These are not documented. Should be done. Will do.

> A few links on the readme leads to nowhere/git.io page's 404 error page.

Whoooops. Will fix. Thanks!

simonmicro, 2025-01-11:
> 2.2.3: Fitness watch face is too powerful for the ESP32, Holding button 3 to enter app drawer stutters as it increases the bubble size.

Hehe, no it is not. Just the UI lib is crap. Take a look how we render circles. Then look how rounded arcs are rendered. It works, but is so slow and wasteful... The ESP32 is so powerful it can even do alpha blending and pretty good framerates on even much bigger screens (e.g. 800x420px), but our GFX is not there (yet).

simonmicro, 2025-01-11:
> 2.1: Watch faces affecting GUI. Large button text until navigating with button, then small overall text:

Can you more clearly describe what you mean? The Numerals bug I've found and fixed, but I do not see what you mean with this.

simonmicro, 2025-01-11:
> 4: If i set firmware to dd/mm/yyyy as i am used to, watch will not boot up at all after flashing. Just black screen bootloop.

Wait. How did you do that? The firmware does not support this... Take a look into the bootlogs via serial, I assume there will be quite some complaining going on about this.

simonmicro, 2025-01-11:
Regarding your feature requests...

>NEW:
> Together with setting page for "wake by lift", add a "automatic calibration submenu so that it can for instance click "calibrate", and it will ask you to flick your wrist 3-5 times to get an average/range of gram force user put into the flick, and save that to the calibration file. And maybe a third page that will output the motion numbers from last calibration, so that users can copy and paste those to a "default user motion data" to a file in the main firmware to auto apply the calibration to future additions if user wants to add/remove apps/functons.

Nice idea for an app, please open a new dedicated issue, as this issue is already random enough (this will also encourage discussion of the final app).

> Feature request in firmware to have a toggleable option in firmware.h to make 2 partitions upon flashing. One for the main firmware, second one for battery and accelerometer calibration data.
> And make a second toggle for dual partition that enables if "dual partition" option is enabled, which will be a yes/no option to retain current second partition data for battery and accelerometer data. That way you don't need to recalibrate/copy paste new data every time.

...the current partition layout is configured using the platformio csv for `min_spiffs.csv` -> 2x app, 1x nvs, 0x spiffs. The NVS is ready for you to use to store such data. Take a look into `Preferences.h` / `src/hal/power.cpp` (`powerPreferences`) to learn more.

> Old:
Documentation on default key binds for zooming in/out with buttons, increase/decrease brightness, set watch face as default. As those are missing.
A few links on the readme leads to nowhere/git.io page's 404 error page.

I added a toast about this and will drop a sentence into the README, but after thinking about this, I do not really see much need for this. Beside the set-default-watchface-button, the other ones are fairly self-explanatory.

DuckersMcQuack, 2025-01-11:
> > 2.1: Watch faces affecting GUI. Large button text until navigating with button, then small overall text:
> 
> Can you more clearly describe what you mean? The Numerals bug I've found and fixed, but I do not see what you mean with this.

Aye. When you enter app drawer,  "category" text top right,  "app" bottom right and "ok" bottom left appear very large, as soon as you press top right button, "ok" becomes very small.  Unless that's intentional?

> Wait. How did you do that? The firmware does not support this... Take a look into the bootlogs via serial, I assume there will be quite some complaining going on about this.

I didn't do that, that's why i report it resulting in a "bootloop", as i hoped i'd be able to use the date formatting we use in norway, which is dd/mm/yyyy, but apparently that won't work :P

> > 2.1: Watch faces affecting GUI. Large button text until navigating with button, then small overall text:
> 
> Can you more clearly describe what you mean? The Numerals bug I've found and fixed, but I do not see what you mean with this.

Aye. When you enter app drawer,  "category" text top right,  "app" bottom right and "ok" bottom left appear very large, as soon as you press top right button, "ok" becomes very small.  Unless that's intentional?

simonmicro, 2025-01-12:
> Aye. When you enter app drawer, "category" text top right, "app" bottom right and "ok" bottom left appear very large, as soon as you press top right button, "ok" becomes very small. Unless that's intentional?

Oh yes. It is the "tutorial", so the user can more easily read it for the first time. But using deep-sleep will reset this flag always, so you may see it more often.

> I didn't do that, that's why i report it resulting in a "bootloop", as i hoped i'd be able to use the date formatting we use in norway, which is dd/mm/yyyy, but apparently that won't work :P

This format is going to be added in #423 

> Aye. When you enter app drawer, "category" text...

_duplicate_

closed 2025-01-14 by simonmicro

DuckersMcQuack, 2025-01-18:
> > Aye. When you enter app drawer, "category" text top right, "app" bottom right and "ok" bottom left appear very large, as soon as you press top right button, "ok" becomes very small. Unless that's intentional?
> 
> Oh yes. It is the "tutorial", so the user can more easily read it for the first time (and yes, navigation may not be intuitive, but is the fastest I could do with only three buttons). But using deep-sleep will reset this flag always, so you may see it more often.
>
Navigation is how i expected it to be :) As the way the system has tools to say long/extra long press can be used as "macros" for lots of more control for the watch :P So it has a lot more going for it than just what it has now.
> This format is going to be added in [#423](https://github.com/Open-Smartwatch/open-smartwatch-os/pull/423)
Nice :)

And newest "bug" on the master branch. when i wake the watch, screen lights up the backlighting, goes dark again, then lights up backlighting, then watch face comes on. So there seems to be something in the middle there holding it back for a second.

PR #423 (2025-01-12): A few bugs and minor issues
- [x] Fixed select button bug on analog+fitness watchface
- [x] Add note how to use default buttons
- [x] Added "dd/mm/yyyy" as date format
- [x] Toast on default watchface change
- [x] Also reset fonts upon redraw
- [x] Interaction on unlocked console also counts as user interaction
- [x] Reimplemented display+bl init order to prevent flashing

Closes #422 (mostly)