Open-Smartwatch/open-smartwatch-os #431: [develop commit 9478a8e bug]

DuckersMcQuack, 2025-01-17:
Bug: If you enter wifi, connect to it, then go back to app drawer, it will bug out by keeping the information within wifi on top of app drawer as well. 

[image]

[image]

DuckersMcQuack, 2025-01-17:
Tested newest master. This issue was indeed solved with that one's latest.

But master has another persisting bug that the latest develop also had.

"IP" text is blinking. 
And question regarding the wifi page. What's the significance of the screen sometimes "blinking slow", as in whole screen gets a digital black curtain that pans in from the right, fills the screen, then the watch pops image back up?

simonmicro, 2025-01-18:
Do you see the red memory icon at the top? The system is in low-memory mode because wifi is using too much RAM. This disables the display buffer, so you have to wait until a full refresh is performed. This is a feature and not a bug - but you are right, we'll need to document this better.

DuckersMcQuack, 2025-01-18:
So wifi alone takes up all the ram more or less? Or can some stuff that just sit in ram be offloaded, and instead use extra time to load after you're done?

And how much ram does bluetooth use? If bluetooth is lighter, for instance hook it up to your phone with a simple "connect only to read system time" for instance.

simonmicro, 2026-09-27:
The problem is the webserver, as its HTTP buffers eat a lot. But nowadays I think there is also another issue of processing this fixed on the secondary core, on which also the kernel resides - so with heavy lifting done during web-requests, the kernel starves due to no execution time and FreeRTOS performing no preemptive scheduling, so its watchdog triggers. Hence, the low-memory mode is maybe a bit aggressive, but also an accidental way of preventing crashes with the latter bug (I suspect nowadays).